RoboCup-SSL/grSim #25: Build `vartypes` inline

mani-monaj, 2016-11-09:
This is the official git repo: https://github.com/szi/vartypes

mani-monaj, 2016-11-09:
I've made some progress on this. The following change will build `vartypes` inline (assuming `vartypes` is copied/symlinked/git-submoduled within the `grSim` folder). There are two issues with this approach:

1. `vartypes` is built as a dynamic library by default (not configurable). Even after a successful build, `grSim` would not run unless the folder that contains the resulting shared library is added to `LD_LIBRARY_PATH`.
2. The include directory structure of `vartypes` and the location of its header files differ between its source and install tree. 

```cmake
add_subdirectory(vartypes EXCLUDE_FROM_ALL)                                     
list(APPEND libs vartypes)                                                      
set_target_properties(vartypes PROPERTIES AUTOMOC False)                        
#find_package(VarTypes REQUIRED)                                                
#include_directories(${VARTYPES_INCLUDE_DIRS})                                  
#list(APPEND libs ${VARTYPES_LIBRARIES})
```

mani-monaj, 2016-11-10:
@mahi97 I could make `vartypes` to build inline (tested only in `Linux`). The code is in the following branch:

https://github.com/mani-monaj/grSim/tree/build-fixes

I've also added some comments to the cmake file to list the issues. To use this branch you need to do `git clone --recursive` or `git submodule update`. Could you please take a look at tell me what you think? Also can you please test it on Mac?

mahi97, 2016-11-10:
Thanks for your complete description, i will test it as soon as possible.

mahi97, 2016-11-10:
I test this branch in macOS and as you know because there's no tr1 (c++ technical report1) in macOS anymore and unfortunately vartypes use share pointers from <tr1/memory> ...  so I thinks maybe it's better to choose another repo to submodule that only for macOS, or just build vartypes and put a binary library inside project for macOS users.

mahi97, 2016-11-11:
It's work perfectly fine on Linux, although, I don't think it's necessary to put `vartypes_example` in `./bin` folder.
if anyone want to test this branch, please don't forget to initial sub-module first by  `git submodule init`

mani-monaj, 2016-11-14:
I am going to push this to milestone 2.1 for now, since the solution is a bit hacky and will break any effort on defining INSTALL rules or packaging.

jgkamat, 2018-01-21:
@mani-monaj, I just wanted to let you know that we actually developed this in parallel (sorry for not letting you know about it), and we have a version which works for us on our [robojackets mods branch](https://github.com/RoboJackets/grSim/tree/robojackets_mods). I'm not sure if it would work on mac os, or how similar it is to the existing attempt. 

If you like it, I'd be happy to factor it out into it's own branch and make a PR, but I'm not sure how useful that would be to you. :)

PR #93 (2019-09-05): Build vartypes inline if find_package fails
👋 again! Please let me know of any feedback you've got on this. I know @mahi97 had a branch to do this, and I actually did this once (a different way) on https://github.com/RoboJackets/grSim, but I decided to start fresh and try an approach with externalproject. 

I think this should be a pretty safe way of doing it and should make building much easier.

- [x] Test install rules
    - [x] Linux - works
    - [x] MacOS - works when installing to /usr/local/bin, building the .app doesn't work (but it wasn't working before this change either)
- [x] Remo [... cut]
closed 2019-09-05 by mahi97