RoboCup-SSL/grSim #34: grsim.xml ?

samedcildir, 2017-01-01:
There is `world=VarXML::read(world,(QDir::homePath() + QString("/.grsim.xml")).toStdString());` line in configwidget.cpp line 130.

Because of that line, BallMass becomes 0 and the program gives `ODE INTERNAL ERROR 1: assertion "dMassCheck(mass)" failed in dBodySetMass()` error.

What is grsim.xml? and is that line really necessary?

mahi97, 2017-01-01:
I don't get this error on my machine ... .
Can you give me the specific of your PC and the way you build grSim and dependencies ?

mahi97, 2017-01-01:
The grsim.xml save the values of Vartypes widget that contain the parameters of geometry, physics and communication.

It used to save data that someone configured for grSim and load it on the next time that grSim is running.

If ballMass becomes 0 it's probably because in grSim.xml file 0 is saved for ballMass,
I think you can change it manually or delete the grsim.xml and let grsim make another one with default values.

samedcildir, 2017-01-02:
I'm using Ubuntu 14.04, I followed the instructions on the GitHub for build.

grsim.xml file looks ok. Ball mass is set to 0.043 but for some reason VarXML::read cannot read it

mahi97, 2017-01-02:
Can you check and make sure that ode is installed in `double-precision` mode or not ?

samedcildir, 2017-01-02:
I found the error. When it writes the doubles it writes with dot(.) such as `0.043`. But for some reason it requires comma(,) when its reading such as `0,043`.

mahi97, 2017-01-02:
Interesting i don't have this problem on mine!
Do you use English for your regions and language too ?

mani-monaj, 2017-01-09:
@samedcildir It's most likely a `locale` problem. I think your system's `locale` uses `,` as the decimal separation character. The easiest (but not the most convenient) is to set your `locale` to `en_US.utf8` or similar. The long term solution would to fix this issue in *VarTypes* (needs confirmation though).

samedcildir, 2017-01-28:
Sorry I forgot to respond. My systems language is English but probably my locale is `tr_TR` so I guess if I change my locale the problem will be solved.

closed 2017-01-29 by mahi97