RoboCup-SSL/grSim #35: Build Grsim On Travis(mac)

mahi97, 2017-01-02:
As brew removed qt4 package and i can't find any other repo that contain it. we need another way to install qt4 inside travis.

mahi97, 2017-01-02:
@mani-monaj I test macports and it contain qt4-mac that is qt4.8.7.
unlike `brew` that is installed by default on travis, we should install `port` manually.
if you agree with this solution i can work on it.
Otherwise i think migration to qt5 is another option but it need lots of changes in cmake file and a little in source code i guess.

mani-monaj, 2017-01-09:
@mahi97 Thanks for looking into a solution. 

Instead of using `macport`, what do you think about using this? https://github.com/cartr/homebrew-qt4

mahi97, 2017-01-09:
I tried this (https://github.com/mahi97/grSim/blob/qt4/.travis.yml) but the problem is travis use El Capitan and it gets timeout in make step of qt4.8.7 so i think we should wait until they close this issue
https://github.com/cartr/homebrew-qt4/issues/8

mani-monaj, 2017-01-09:
I suggest that we disable the build on TravisCI for OSX at the moment and wait for a better solution to emerge from the community. 

How difficult it is in your opinion to migrate the codebase to Qt 5.x?

mahi97, 2017-01-09:
I so Agree with this :)
Actually it's possible to build and use grSim on mac if you can install qt4 on your machine, the problem is just with travis so we can disable it for now and wait for better solution

mani-monaj, 2017-01-09:
Thanks @mahi97. Could you please make the required changes to both the travis and documentation files? I will create a new issue to migrate to Qt 5.x .

mahi97, 2017-01-09:
I thinks i found a new solution (in cartr/homebrew-qt4#8) so i will try this one and put the result here i really hope that this solution work well.

closed 2017-01-10 by mahi97