RoboCup-SSL/grSim #38: Gitter ?

mahi97, 2017-01-10:
@mani-monaj 
I think it's good idea that we have some chat about to how solve issues on Gitter if you're agree i will setup Gitter for grSim

mani-monaj, 2017-02-08:
@mahi97 It's a great idea! 👍

mahi97, 2018-01-15:
@g3force Permission to access public repo should be granted for Gitter to let me create a room.
Could you grant this permission for RoboCup-SSL?

g3force, 2018-01-15:
I am not able to give the permission either. @joydeep-b has to do it.

joydeep-b, 2018-01-15:
Could you explain what features it would bring, that are not already available on Github? Issues on Github (such as this) seem to be a good way to raise issues, discuss the solution, and then tie them to PRs for the solution.

mahi97, 2018-01-15:
Gitter is a chat and networking platform that will help us to grow and connect the community through messaging, content and discovery, also a suitable place for asking any kind of question about the project, not only problems that can be issued.
I think a chat platform is a great chance for teams to communicate with each other easier, and It will lead to a more active SSL community.
I can't see a really necessary need to have a chat platform, but I believe it will be a great enhancement for the community.

g3force, 2018-01-16:
I like the idea of a community chat and if gitter brings us such thing, I vote for it. I also don't see any reason why we should not try it out.