RoboCup-SSL/grSim #41: ODE 0.13 Ubuntu 16.04

Crafter76, 2017-03-13:
At the first start, when I made files there were no problems with ./grsim.
But at the second start, I caught

crafter76@crafter76-dexp:~/grsim_ws/grSim/bin$ ./grsim 

ODE Message 2: mass must be > 0 in dMassCheck() [mass.cpp:49]

ODE Message 2: mass must be > 0 in dMassCheck() [mass.cpp:49]

ODE INTERNAL ERROR 1: assertion "dMassCheck(mass)" failed in dBodySetMass() [ode.cpp:485]

And nothing have helped me to fix this error.

mahi97, 2017-03-14:
Are you sure that you'r using ODE with double precision ?

pabloreyesrobles, 2017-03-20:
I'm having the same issue. I did "./configure --enable-double-precision --with-trimesh=opcode --enable-new-trimesh --enable-shared -enable-release" before installing, as I saw in other threads assuming that will solve the problem, but I'm still getting the error.
I have Ubuntu 16.04 and recently installed ODE 0.13.

mahi97, 2017-03-20:
As I personally tested grSim only on Ubuntu 14.04 and macOS i'm not sure about it, but is seems, as it's happened for the second time, that config files (~.xml) which made by vartypes from the first time cause this error. maybe because language setting or any other related settings.
I will test grSim on 16.04 ASAP.

mahi97, 2017-03-24:
I test it on 16.04 lately and i hadn't any problem, can you share your *.xml files here ?

bermeom, 2017-03-28:
Yes, the first time grSim was working, but later emerge this problem, and I thought that problem was GCC or Operative system (Ubuntu 16.04), then I tested with Ubuntu 14.04 but was same.

Anamaria11, 2017-04-03:
Hola,
What .xml files you need? I dont see .xml files in my grSim folder

pabloreyesrobles, 2017-04-04:
@mahi97  I erased the .grsim.xml file on home folder, and I was able to run the simulator again, but got this:

XML Parsing error inside file '/home/pablo/.grsim.xml'.
Error: File not found
At line 0, column 0.

I also got this the first time I ran the the simulator after installed it but I think wasn't important, what could it be? Is it our problem? Sharing my .xml by the way.

[grsim.xml.zip](https://github.com/mani-monaj/grSim/files/894072/grsim.xml.zip)

Anamaria11, 2017-04-05:
Error is the second time that you run grsim. My .xml file contains:
<VarXML><VarList name="Geometry"><VarList name="Field"><VarDouble name="Line Thickness" minval="" maxval="">
				0.01
			</VarDouble><VarDouble name="Length" minval="" maxval="">
				9
			</VarDouble><VarDouble name="Width" minval="" maxval="">
				6
			</VarDouble><VarDouble name="Radius" minval="" maxval="">
				0.5
			</VarDouble><VarDouble name="Defense Radius" minval="" maxval="">
				1
			</VarDouble><VarDouble name="Defense Stretch" minval="" maxval="">
				0.5
			</VarDouble><VarDouble name="Free Kick Distanse From Defense Area" minval="" maxval="">
				0.7
			</VarDouble><VarDouble name="Penalty radius" minval="" maxval="">
				0.5
			</VarDouble><VarDouble name="Penalty line length" minval="" maxval="">
				0.35
			</VarDouble><VarDouble name="Penalty point" minval="" maxval="">
				0.95
			</VarDouble><VarDouble name="Margin" minval="" maxval="">
				0.25
			</VarDouble><VarDouble name="Referee margin" minval="" maxval="">
				0.455
			</VarDouble><VarDouble name="Wall thickness" minval="" maxval="">
				0.05
			</VarDouble><VarDouble name="Goal thickness" minval="" maxval="">
				0.02
			</VarDouble><VarDouble name="Goal depth" minval="" maxval="">
				0.2
			</VarDouble><VarDouble name="Goal width" minval="" maxval="">
				1
			</VarDouble><VarDouble name="Goal height" minval="" maxval="">
				0.16
			</VarDouble></VarList><VarList name="Ball"><VarDouble name="Radius" minval="" maxval="">
				0.0215
			</VarDouble></VarList><VarStringEnum name="Blue Team">
			RoboIME2012
			<VarString name="0">
				Parsian
			</VarString><VarString name="1">
				ParsianNew
			</VarString><VarString name="2">
				RoboIME2012
			</VarString><VarString name="3">
				Parsian
			</VarString><VarString name="4">
				ParsianNew
			</VarString><VarString name="5">
				RoboIME2012
			</VarString></VarStringEnum><VarStringEnum name="Yellow Team">
			Parsian
			<VarString name="0">
				Parsian
			</VarString><VarString name="1">
				ParsianNew
			</VarString><VarString name="2">
				RoboIME2012
			</VarString><VarString name="3">
				Parsian
			</VarString><VarString name="4">
				ParsianNew
			</VarString><VarString name="5">
				RoboIME2012
			</VarString></VarStringEnum></VarList><VarList name="Physics"><VarList name="World"><VarDouble name="Desired FPS" minval="" maxval="">
				65
			</VarDouble><VarBool name="Synchronize ODE with OpenGL">
				false
			</VarBool><VarDouble name="ODE time step" minval="" maxval="">
				0.016
			</VarDouble><VarDouble name="Gravity" minval="" maxval="">
				9
			</VarDouble></VarList><VarList name="Ball"><VarDouble name="Ball mass" minval="" maxval="">
				0.043
			</VarDouble><VarDouble name="Ball-ground friction" minval="" maxval="">
				0.05
			</VarDouble><VarDouble name="Ball-ground slip" minval="" maxval="">
				1
			</VarDouble><VarDouble name="Ball-ground bounce factor" minval="" maxval="">
				0.5
			</VarDouble><VarDouble name="Ball-ground bounce min velocity" minval="" maxval="">
				0.1
			</VarDouble><VarDouble name="Ball linear damping" minval="" maxval="">
				0.004
			</VarDouble><VarDouble name="Ball angular damping" minval="" maxval="">
				0.004
			</VarDouble></VarList></VarList><VarList name="Communication"><VarString name="Vision multicast address">
			224.5.23.2
		</VarString><VarInt name="Vision multicast port" minval="" maxval="">
			10020
		</VarInt><VarInt name="Command listen port" minval="" maxval="">
			20011
		</VarInt><VarInt name="Blue Team status send port" minval="" maxval="">
			30011
		</VarInt><VarInt name="Yellow Team status send port" minval="" maxval="">
			30012
		</VarInt><VarInt name="Sending delay (milliseconds)" minval="" maxval="">
			0
		</VarInt><VarInt name="Send geometry every X frames" minval="" maxval="">
			120
		</VarInt><VarList name="Gaussian noise"><VarBool name="Noise">
				false
			</VarBool><VarDouble name="Deviation for x values" minval="" maxval="">
				3
			</VarDouble><VarDouble name="Deviation for y values" minval="" maxval="">
				3
			</VarDouble><VarDouble name="Deviation for angle values" minval="" maxval="">
				2
			</VarDouble><VarBool name="Vanishing">
				false
			</VarBool></VarList><VarList name="Vanishing probability"><VarDouble name="Blue team" minval="" maxval="">
				0
			</VarDouble><VarDouble name="Yellow team" minval="" maxval="">
				0
			</VarDouble><VarDouble name="Ball" minval="" maxval="">
				0
			</VarDouble></VarList></VarList></VarXML>

Anamaria11, 2017-04-05:
@mahi97 Hello,
Possibly I have a solution. I think there are issues with this vartypes file, because: 
I deleted vartypes and I have reinstalled it and i had this error:
`[ 41%] Building CXX object CMakeFiles/vartypes.dir/vartypes/gui/moc_VarItem.o
In file included from /tmp/vartypes/./vartypes/VarTypesFactory.h:37:0,
                 from /tmp/vartypes/./vartypes/VarTypesInstance.h:17,
                 from /tmp/vartypes/./vartypes/VarTypes.h:42,
                 from /tmp/vartypes/build/vartypes/gui/../../../vartypes/gui/VarItem.h:25,
                 from /tmp/vartypes/build/vartypes/gui/moc_VarItem.cxx:9:
/tmp/vartypes/./vartypes/primitives/VarAny.h:27:45: fatal error: boost/archive/binary_iarchive.hpp: No existe el archivo o el directorio
compilation terminated.
CMakeFiles/vartypes.dir/build.make:444: fallo en las instrucciones para el objetivo 'CMakeFiles/vartypes.dir/vartypes/gui/moc_VarItem.o'
make[2]: *** [CMakeFiles/vartypes.dir/vartypes/gui/moc_VarItem.o] Error 1
CMakeFiles/Makefile2:67: fallo en las instrucciones para el objetivo 'CMakeFiles/vartypes.dir/all'
make[1]: *** [CMakeFiles/vartypes.dir/all] Error 2
Makefile:127: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2
`
I have had to install vartypes from google like this:
- I downloaded from page: https://code.google.com/archive/p/vartypes/downloads (vartypes-0.7.tar.gz), and then I did this:
`tar xfz vartypes-0.7.tar.gz
cd vartypes-0.7
mkdir build && cd build
cmake ..
make
sudo make install`

pabloreyesrobles, 2017-04-08:
@Anamaria11 It worked for me! Thank you a lot. I'm fighting with the simulator right now but I have no issues now running it.

mahi97, 2017-04-08:
Sorry for late response i was too busy about IranOpen.
I think any error related to xml files can only came from vartypes xml file, but grSim currently work with vartypes 0.8 and vartypes from google code is version 0.7 so I think change form version 0.7 to 0.8 may cause this issue. but I never get this error though so i still thinking that the problem is the way that vartypes save and load data form xml files.

jellezult, 2017-10-21:
Many of our team had this problem. Eventually it turned out to be because their Ubuntu was configured for the dutch format, where the comma is used as a decimal. The grsim xml uses dots as decimal, so there it probably goes wrong when ODE reads these. So make sure your system uses dots as decimal.

bermeom, 2017-10-21:
I solved this problem using a Docker container (it is easy if you have any problem it just restart container) look my repository: https://github.com/bermeom/grSim-docker or https://hub.docker.com/r/bermeom/grsim/

renatoosousa, 2018-04-29:
thanks @Anamaria11, it work for me too

edgarcamilocamacho, 2018-04-30:
(Ubuntu 18.04 LTS and 16.04 LTS) I compiled and installed ODE with the "--enable-double-precision" flag but I got the same error, then, I tried to apply the @Anamaria11 's solution (installing vartypes 0.7) but I couldn't because of problems with c++11, but I found that, if I close grSim, open ~/.grsim.xml and change all the dots by commas (except by "﻿<?xml version="1.0" encoding="utf-8"?>" and "<VarString name="Vision multicast address"> 224.5.23.2 </VarString>"), it works. But I have to do that every time I want to open grSim.

edgarcamilocamacho, 2018-04-30:
It seems that the problem is like this:
https://stackoverflow.com/questions/13489940/stupid-bug-with-fscanf-comma-instead-of-point
I solved the problem setting the following environment variables before executing grsim:
$ export LANG=en_US.UTF-8
$ export LC_CTYPE=en_US.UTF-8
$ export LC_ALL=en_US.UTF-8

renatoosousa, 2018-05-02:
@edgarcamilocamacho This worked for me too.

mahi97, 2019-01-04:
Please test this PR #84.
I Hope this fixes the issue.

closed 2019-01-05 by mahi97