RoboCup-SSL/grSim #64: Compiling issue on MacOS 10.13

scien, 2018-03-13:
[x] dependencies installed
[x] osx-fix branch of lordhippo's vartype repo built and installed
[ ] compile grSim

`cmake ..`

```
bryant@Bryants-MacBook-Pro:~/dev/robocup/grSim/build> cmake ..
-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - not found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - found
-- Looking for QT_MAC_USE_COCOA
-- Looking for QT_MAC_USE_COCOA - found
-- Found Qt4: /usr/local/bin/qmake (found version "4.8.7")
-- Found ODE: /usr/local/include
-- Found VARTYPES: /usr/local/include
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Found Protobuf: /usr/local/lib/libprotobuf.dylib (found version "3.5.1")
-- Configuring done
CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0071 is not set: Let AUTOMOC and AUTOUIC process GENERATED files.
  Run "cmake --help-policy CMP0071" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the GENERATED source file(s):

    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_detection.pb.h"
    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_geometry.pb.h"
    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_wrapper.pb.h"
    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_refbox_log.pb.h"
    "/Users/bryant/dev/robocup/grSim/build/grSim_Replacement.pb.h"
    "/Users/bryant/dev/robocup/grSim/build/grSim_Commands.pb.h"
    "/Users/bryant/dev/robocup/grSim/build/grSim_Packet.pb.h"
    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_detection.pb.cc"
    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_geometry.pb.cc"
    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_wrapper.pb.cc"
    "/Users/bryant/dev/robocup/grSim/build/messages_robocup_ssl_refbox_log.pb.cc"
    "/Users/bryant/dev/robocup/grSim/build/grSim_Replacement.pb.cc"
    "/Users/bryant/dev/robocup/grSim/build/grSim_Commands.pb.cc"
    "/Users/bryant/dev/robocup/grSim/build/grSim_Packet.pb.cc"
    "/Users/bryant/dev/robocup/grSim/build/qrc_textures.cxx"

  from processing by AUTOMOC.  If any of the files should be processed, set
  CMP0071 to NEW.  If any of the files should not be processed, explicitly
  exclude them by setting the source file property SKIP_AUTOMOC:

[... 77 lines cut ...]
[ 56%] Building CXX object CMakeFiles/grSim.dir/src/physics/pray.cpp.o
[ 58%] Building CXX object CMakeFiles/grSim.dir/src/net/robocup_ssl_server.cpp.o
[ 60%] Building CXX object CMakeFiles/grSim.dir/src/net/robocup_ssl_client.cpp.o
[ 62%] Building CXX object CMakeFiles/grSim.dir/src/sslworld.cpp.o
[ 64%] Building CXX object CMakeFiles/grSim.dir/src/robot.cpp.o
[ 66%] Building CXX object CMakeFiles/grSim.dir/src/configwidget.cpp.o
[ 68%] Building CXX object CMakeFiles/grSim.dir/src/statuswidget.cpp.o
[ 70%] Building CXX object CMakeFiles/grSim.dir/src/logger.cpp.o
[ 72%] Building CXX object CMakeFiles/grSim.dir/src/robotwidget.cpp.o
[ 74%] Building CXX object CMakeFiles/grSim.dir/src/getpositionwidget.cpp.o
[ 76%] Building CXX object CMakeFiles/grSim.dir/grSim_autogen/mocs_compilation.cpp.o
[ 78%] Linking CXX executable ../bin/grSim.app/Contents/MacOS/grSim
Undefined symbols for architecture x86_64:
  "_ccdFirstDirDefault", referenced from:
      ccdCollide(dxGeom*, dxGeom*, int, dContactGeom*, int, void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*)) in libode.a(collision_libccd.o)
  "_ccdMPRIntersect", referenced from:
      ccdCollide(dxGeom*, dxGeom*, int, dContactGeom*, int, void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*)) in libode.a(collision_libccd.o)
  "_ccdMPRPenetration", referenced from:
      ccdCollide(dxGeom*, dxGeom*, int, dContactGeom*, int, void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*)) in libode.a(collision_libccd.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../bin/grSim.app/Contents/MacOS/grSim] Error 1
make[1]: *** [CMakeFiles/grSim.dir/all] Error 2
make: *** [all] Error 2
```

scien, 2018-03-13:
one note on the install: the docs said to `brew install freecad/freecad/qt` but that installs qt@5.6.  I was able to get v4 using `brew install cartr/qt4/qt`

mahi97, 2018-03-13:
It seems that the problem is about ODE, the physic engine, version.
`brew` recently updates the version of ode from 0.13 to 0.15.2, I believe grsim works fine on ODE 0.13 and problem is grsim compatibility with the new versions.
So I suggest you build ode-0.13 from [source](https://sourceforge.net/projects/opende/files/) and test it again.
note that  
1. remove ode installed by brew before installing the new one.
2. configure the project with this arguments: `./configure --enable-double-precision --with-trimesh=opcode --enable-new-trimesh --enable-shared`

mahi97, 2018-03-13:
I'll work on  compatibility for this version.

scien, 2018-03-13:
the build was successful after switching to ode-0.13.  The client app runs, but the grSim app crashes on startup.

```
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff5ddd9e3e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff5df18150 pthread_kill + 333
2   libsystem_c.dylib             	0x00007fff5dd36312 abort + 127
3   QtCore                        	0x00000001187d426b qt_message_output(QtMsgType, char const*) + 165
4   QtCore                        	0x00000001187d44f8 qt_message(QtMsgType, char const*, __va_list_tag*) + 404
5   QtCore                        	0x00000001187d3e4d qFatal(char const*, ...) + 139
6   QtGui                         	0x0000000117cf9305 QWidgetPrivate::QWidgetPrivate(int) + 639
7   QtGui                         	0x0000000117cfa254 QWidget::QWidget(QWidget*, QFlags<Qt::WindowType>) + 52
8   QtGui                         	0x0000000117c83b93 QDesktopWidget::QDesktopWidget() + 31
9   QtGui                         	0x0000000117cc5d64 QApplication::desktop() + 50
10  QtGui                         	0x0000000117c7fc3e flipPoint(CGPoint const&) + 32
11  QtGui                         	0x0000000117c7b05a -[QCocoaWindowDelegate windowDidMove:] + 134
12  com.apple.CoreFoundation      	0x00007fff3638cbbc __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
13  com.apple.CoreFoundation      	0x00007fff3638caaa _CFXRegistrationPost + 442
14  com.apple.CoreFoundation      	0x00007fff3638c7f2 ___CFXNotificationPost_block_invoke + 50
15  com.apple.CoreFoundation      	0x00007fff3634a670 -[_CFXNotificationRegistrar find:object:observer:enumerator:] + 1664
16  com.apple.CoreFoundation      	0x00007fff363497a3 _CFXNotificationPost + 595
17  com.apple.Foundation          	0x00007fff3842e467 -[NSNotificationCenter postNotificationName:object:userInfo:] + 66
18  com.apple.AppKit              	0x00007fff34222e88 _NSWindowSendWindowDidMove + 85
19  com.apple.AppKit              	0x00007fff33a1b6a6 -[NSWindow _setFrameCommon:display:stashSize:] + 3216
20  com.apple.AppKit              	0x00007fff33a1aa09 -[NSWindow _setFrame:display:allowImplicitAnimation:stashSize:] + 222
21  com.apple.AppKit              	0x00007fff33ae809a -[NSWindow setFrameOrigin:] + 456
22  QtGui                         	0x0000000109142429 QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) + 861
23  QtGui                         	0x00000001091d8096 QWidget::resize(QSize const&) + 98
24  QtGui                         	0x00000001091d2e14 QWidget::setMinimumSize(int, int) + 188
25  QtGui                         	0x00000001091bb240 QLayout::activate() + 520
26  QtGui                         	0x00000001091d53fe QWidget::adjustSize() + 66
27  QtGui                         	0x00000001091416e3 QWidget::setWindowState(QFlags<Qt::WindowState>) + 749
28  QtGui                         	0x00000001091d1e9c QWidget::showMaximized() + 68
29                                	0x0000000108c622be MainWindow::MainWindow(QWidget*) + 10670
30                                	0x0000000108c63ead MainWindow::MainWindow(QWidget*) + 29
31                                	0x0000000108c5ed52 main + 274
32  libdyld.dylib                 	0x00007fff5dc8a115 start + 1
```

```
Binary Images:
       0x108c27000 -        0x108fa7ff7 + (1.0.0a2 - ???) <472F90BC-E093-3861-A446-C2FE9E3D7786> /Users/USER/*/grSim.app/Contents/MacOS/grSim
       0x109015000 -        0x1090d7ff7 +QtOpenGL (4.8 - 4.8.7) <47BFC3B2-2D9D-381F-ABE4-68F216383703> /Users/USER/*/grSim.app/Contents/Frameworks/QtOpenGL.framework/Versions/4/QtOpenGL
       0x109121000 -        0x1099b6fff +QtGui (4.8 - 4.8.7) <AEDBE9CD-EE44-317A-8B5C-F645A94ABDCD> /Users/USER/*/grSim.app/Contents/Frameworks/QtGui.framework/Versions/4/QtGui
       0x109c96000 -        0x109d9cfff +QtNetwork (4.8 - 4.8.7) <F3D9C3A9-F390-362E-A9C3-1E8DF57D283F> /Users/USER/*/grSim.app/Contents/Frameworks/QtNetwork.framework/Versions/4/QtNetwork
       0x109e09000 -        0x10a07afff +QtCore (4.8 - 4.8.7) <A1663CC0-450F-36EB-BBA6-8A5CB0AA5C33> /Users/USER/*/grSim.app/Contents/Frameworks/QtCore.framework/Versions/4/QtCore
       0x10a12b000 -        0x10a1f8fff +libode.3.dylib (0) <51663AFD-6344-3F63-AB39-40088F3109AB> /Users/USER/*/grSim.app/Contents/MacOS/libode.3.dylib
       0x10a26d000 -        0x10a2bdffb +libvartypes.0.7.0.dylib (0) <377D3ADE-E52C-3637-8338-2941E094734D> /Users/USER/*/grSim.app/Contents/MacOS/libvartypes.0.7.0.dylib
       0x10a2fd000 -        0x10a42fff7 +libprotobuf.15.dylib (0) <7126DB9C-1EE3-36D1-B366-9047D43A8A3C> /Users/USER/*/grSim.app/Contents/MacOS/libprotobuf.15.dylib
       0x10a537000 -        0x10a578fff +libssl.1.0.0.dylib (0) <CB91FB11-F59B-33C4-8D22-9E4917F6CF03> /Users/USER/*/grSim.app/Contents/MacOS/libssl.1.0.0.dylib
       0x10a598000 -        0x10a708c87 +libcrypto.1.0.0.dylib (0) <98FFE817-09E4-387B-B497-D4E74740AD0C> /Users/USER/*/grSim.app/Contents/MacOS/libcrypto.1.0.0.dylib
       0x10c1d3000 -        0x10c21d98f  dyld (519.2.2) <6695F30B-4E88-3C0B-9867-7D738C44A3E6> /usr/lib/dyld
       0x10c6e8000 -        0x10c6edfff +libqgif.dylib (0) <C7DAC2E4-1EDE-3E24-8D1F-54F4E42C5234> /usr/local/Cellar/qt@4/4.8.7_3/lib/qt4/plugins/imageformats/libqgif.dylib
[... cut ...]
       0x118bdf000 -        0x118c16fff +QtXml (4.8.7) <3C7376A5-725F-333C-BADD-9F9D044EC074> /usr/local/Cellar/qt@4/4.8.7_3/lib/QtXml.framework/Versions/4/QtXml
       0x118c31000 -        0x118c35fff +libqtga.dylib (0) <2DE8CC73-106D-34D6-A2A5-F1431F8513AD> /usr/local/Cellar/qt@4/4.8.7_3/lib/qt4/plugins/imageformats/libqtga.dylib
       0x118c3a000 -        0x118c92ff7 +libqtiff.dylib (0) <F5C9C253-E992-3D93-A9A3-C8B84757F8A5> /usr/local/Cellar/qt@4/4.8.7_3/lib/qt4/plugins/imageformats/libqtiff.dylib
...
```

let me know if you need anything else from the crash log to diagnose

mahi97, 2018-03-15:
I never get this error on my machine and I have done some test and this crash doesn't show up.
Are you sure that your qt is okay?
How did you installed qt?

scien, 2018-03-17:
i believe qt is the issue.  

attempt 1
 - freecad on brew no longer has 4.x (it does have 5.6)
 - qt@5.6 requires a full install of xcode
 - qt@5.6 build is broken on OSX 10.13 xcode 9 (https://bugreports.qt.io/browse/QTBUG-62266)
 - patch was applied in qt@5.9 (https://codereview.qt-project.org/#/c/201546/)
 - installed qt5.10 from source
 - tried to rebuilt grSim, cmake error: "this code requires Qt 4.x'
 - uninstalled qt@5.6

attempt 2
 - tried `brew install cartr/qt4/qt`
 - grSim builds, but then crashes on application start

attempt 3
 - try to build qt@4.8.7 from source
 - get the same build error as with 5.6

what version of qt are you using?  Are you on macOS High Sierra 10.13.3 or on an earlier OS?  Is it feasible to upgrade grSim to qt5?

mahi97, 2018-03-17:
I working On macOS High Sierra 10.13.2 and install qt4 via `brew install cartr/qt4/qt`.
upgrading to Qt5 mostly is done, just fixing `CMakeLists.txt` and build `vartypes` with Qt5 is remained.
The [main repo of vartype](https://github.com/szi/vartypes) is still working with Qt5.
and [this repo](https://github.com/legacy-roboime/vartypes/tree/qt5) is not the last version of vartypes but it's upgraded to Qt5.
maybe the only solution for having grSim with Qt5 is to have our own repo and keep it up to date.

scien, 2018-03-19:
updated vartypes for qt5 (legacy-roboime wasn't working with qt 5.10.1 for me)
https://github.com/scien/vartypes/pull/1/files

updated grSim for qt5
https://github.com/scien/grSim/pull/1/files

unfortunately grSim still crashes on startup. any chance the issue may be related to a library other than qt?

```
Process:               grSim [15314]
Path:                  /Users/USER/*/grSim.app/Contents/MacOS/grSim
Identifier:            ???
Version:               1.0.0a2 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           grSim [15314]
User ID:               501

Date/Time:             2018-03-18 21:23:34.239 -0400
OS Version:            Mac OS X 10.13.3 (17D102)
Report Version:        12
Anonymous UUID:        35600F79-14F4-E10D-F4C2-CF68E0417D55

Sleep/Wake UUID:       D7470087-D3A9-49C8-A9D6-6EF4E8FAD781

Time Awake Since Boot: 81000 seconds
Time Since Wake:       2600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0:
--> 
    __TEXT                 000000010f2c9000-000000010f646000 [ 3572K] r-x/rwx SM=COW  ) [/Users/bryant/dev/robocup/grSim2/bin/grSim.app/Contents/MacOS/grSim]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_c.dylib             	0x00007fff76b1e432 strlen + 18
1                                 	0x000000010f2d6aa5 std::__1::char_traits<char>::length(char const*) + 21
2                                 	0x000000010f300b67 main + 615
3   libdyld.dylib                 	0x00007fff76ace115 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff76c1e562 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff76d5906a _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff76d58c4d start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff76c1e562 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff76d5906a _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff76d58c4d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff76c1e562 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff76d5906a _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff76d58c4d start_wqthread + 13

Thread 4:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	0x00007fff76c1dfca __select + 10
1   com.apple.CoreFoundation      	0x00007fff4f1f659b __CFSocketManager + 635
2   libsystem_pthread.dylib       	0x00007fff76d596c1 _pthread_body + 340
3   libsystem_pthread.dylib       	0x00007fff76d5956d _pthread_start + 377
[... 297 lines cut ...]
MALLOC guard page                   48K       11 
STACK GUARD                       56.0M        6 
Stack                             10.0M        6 
VM_ALLOCATE                         84K       11 
__DATA                            23.4M      256 
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    2588K        2 
__LINKEDIT                       196.3M       27 
__TEXT                           173.9M      254 
__UNICODE                          560K        2 
mapped file                       49.7M       17 
shared memory                     31.2M       12 
===========                     =======  ======= 
TOTAL                            738.2M      663 

Model: MacBookPro14,1, BootROM MBP141.0169.B00, 2 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.43f6
Graphics: Intel Iris Plus Graphics 640, Intel Iris Plus Graphics 640, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x170), Broadcom BCM43xx 1.0 (7.77.37.5.1a3)
Bluetooth: Version 6.0.2f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.1
```

mahi97, 2018-03-19:
Nice work about Qt5, I'm kinda busy about IranOpen competition right now and will check your work and all stuff on macOs from scratch after IranOpen.
For now on I suggest using Ubuntu or Docker for your test.
btw IranOpen will be finished on April 7th.
Also, you can see works done for Docker in #42.
I think you should use `vnc` for share gui from Docker in mac and it's a lot easier.