RoboCup-SSL/grSim #76: -std=c++11 required to build due to protobuf generted code

gwwatkin, 2018-10-08:
On my system (Archlinux, running protobuf 3.6.1 installed from the official repo) build fails  with error messages such as 
```
/home/george/grsim/grSim/build/messages_robocup_ssl_detection.pb.h:285:10: error: ‘constexpr’ does not name a type; did you mean ‘ucontext_t’?
   static constexpr int kIndexInFileMessages =
````
 which really look like a  c++ std version problem.

These methods errors come from the files automatically generated by protobuf.

Changin CMakeLists.txt, the lines

```
# set explicitly the c++ standard to use
set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++03")
```
to
```
# set explicitly the c++ standard to use
set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
```
does indeed solve the problem. With those changes everything works fine, including the demo client in qt.

Maybe this change should be conditional up on some proprety of protobuf (version maybe), but I couldn't figure out why the generated code is c++11.

Since c++11 is widely supported and backwards compatible, would it be fine to change the line to simply c++ 11?

renzenicolai, 2018-10-25:
I experienced the same problem and can confirm that changing the C standard does indeed solve the problem.

Thank you!

mahi97, 2018-10-26:
Thanks.
Solved in #77

closed 2018-10-26 by mahi97