RoboCup-SSL/grSim #87: reset all

amirbayat98, 2019-03-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
when there is more than 1 robot fell down. its hard to click on everyone of them and click reset. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
you can add a reset all button.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kkimurak, 2020-05-23:
Hi, I've implemented such feature. How about this?

[image]

mahi97, 2020-05-23:
It can be closed thanks to #115 #119

closed 2020-05-23 by mahi97