RoboCup-SSL/grSim #89: Field Line coordinate autogeneration is inconsistent with SSL-Vision's generated coordinates

rolfvdhulst, 2019-07-29:
In SSLWorld.cpp the geometry is generated. The centre of each line is corrected by amount of LineThickness, however, in SSL-vision this does not happen. Not sure which one to take, but atleast both GrSim and SSL-vision should be consistent with each other.

mahi97, 2019-07-30:
IMHO, You should try to be robust and more flexible to this configuration at this rate, we experienced robocups with wrong field size!

But you're right, we can at least have an standard about line and area of the field (maybe it's mentioned in rules implicitly).

Anyway, please feel free to send a pull request if ssl technical committee agreed on any of them.

rolfvdhulst, 2019-07-30:
The problem was the rules did not mention a standard implicitly on how the lines should be communicated through vision (e.g. take the 'outside' or the centre of the line), and this makes it easy to misinterpret. I notified the technical committee but still wanted to make this an issue so that users are aware.

g3force, 2021-02-10:
There is an issue for this issue: https://github.com/RoboCup-SSL/ssl-rules/issues/33

I've just created a patch for clarification in the rules. That would mean, we would adapt the behavior in grSim two that of ssl-vision.