RoboCup-SSL/grSim #94: Some links in documents are old

kkimurak, 2019-09-08:
**overview**
Some links in documentations seems to be old.

**detail**
Here is the know list : 
- [README.md (branch : master)](https://github.com/RoboCup-SSL/grSim/blob/master/README.md)
  - Section "Software Requirements"
    - VarTypes repository
      current : ```[VarTypes Library](https://github.com/szi/vartypes)```  
      It must be : ```[VarTypes Library](https://github.com/jpfeltracco/vartypes) forked from [Szi's Vartypes](https://github.com/szi/vartypes)```
    - Qt
      current : ```[Qt4 Development Libraries](https://www.qt.io) version 4.8+```
      It must be : ```[Qt5 Development Libraries](https://www.qt.io)```
- [INSTALL.md (branch : master, develop)](https://github.com/RoboCup-SSL/grSim/blob/develop/INSTALL.md)
  - Section "Linux/Unix Installation"
    - grSim repository
      current : ```$ git clone https://github.com/mani-monaj/grSim.git```
      It must be : ```$ git clone https://github.com/robocup-ssl/grSim.git```

P.S. I've wrote this believing INSTALL.md is correct (because it has recently updated)

mahi97, 2019-09-08:
Good points, feel free to send a pull request for this updates, I would really appriciate 😁👍

kkimurak, 2019-09-08:
(I'm sorry, it seems I added the label by mistake. This is not so serious... )
Thanks, I'll send a pull request soon.

mahi97, 2019-09-08:
No problem, thanks for your efforts 🖐

PR #95 (2019-09-09): Update documents
### Description of the Change

This pull request updates links in some documents.
It contains the change related to dependency updates (VarTypes and Qt5 - see #90)  
and updates of links for grSim repository in the installation procedure for Linux/Unix.
Link for Parsian's we page also updated because old page (wiki.robocup.org) is no longer exists.

Related with the work #93, requirement for CMake also updated from v2.8 to v3.5. 

Also updates for CHANGELOG.md are contains in this pull request.
Update link for commit history, add the [activities for Milestone 2.0](https://gith [... cut]
closed 2019-09-09 by mahi97