RoboCup-SSL/grSim #102: installation error, about the ode

DanielBamboo, 2019-10-01:
```
/usr/lib/libode.a(collision_libccd.o)：在函数‘ccdCollide(dxGeom*, dxGeom*, int, dContactGeom*, int, void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*)) [clone .isra.2] [clone .part.3] [clone .constprop.7]’中：
/home/henryw/ode-0.16/ode/src/collision_libccd.cpp:322：对‘ccdFirstDirDefault’未定义的引用
/home/henryw/ode-0.16/ode/src/collision_libccd.cpp:339：对‘ccdMPRPenetration’未定义的引用
/home/henryw/ode-0.16/ode/src/collision_libccd.cpp:332：对‘ccdMPRIntersect’未定义的引用
```
hello, I just install all these dependences, but there are some errors when I "make" grSim, do I get a wrong ode version ?

mahi97, 2019-10-01:
Are you using osx?

DanielBamboo, 2019-10-01:
I am using Ubuntu 18.04LTS, and sorry I don't understand the two lines of osx commands, could you give me some other suggestions ?

DanielBamboo, 2019-10-01:
there is something wrong with my ode, I did the wrong installation steps

closed 2019-10-01 by DanielBamboo