RoboCup-SSL/grSim #104: Better AABB pruning for robots.

rolfvdhulst, 2019-10-24:
Currently in the collision space of ODE all objects are added to the global collision space. Since there is no collision space per robot ODE is performing collision checks between every seperate object. However, making a collision space per robot and using dSpaceCollide2 (see  http://ode.org/wiki/index.php?title=Manual#Collision_detection_2 ) would make a lot more sense, since then we would only check if two robots collide if they are close enough to do so, without checking the individual pairs of objects between the two robots. Since most situations of gameplay have limited collisions between robots, this could be a significant speed up in the collision checking of grSim (Ignoring static objects; factor 7x*(7x-1)/(x*(x-1)), where x is the amount of robots. This works out to roughly ~49 for 22 robots). Speedup in the collision checking could possibly be even more due to the possibility of usage of a different data structure for the space.

mahi97, 2019-10-24:
It seems so reasonable to me.
I believe besides graphics, most of the process is only for collisions so that this change can increase overall performance significantly.

rolfvdhulst, 2021-02-04:
I built this, but unfortunately it doesn't give any detectable /measurable speedup or performance difference, so I'm closing this issue.

closed 2021-02-04 by rolfvdhulst