RoboCup-SSL/grSim #109: When compiling，err occurs: "Could NOT find ODE" on Ubuntu 18.04 LTS

Droliven, 2020-03-25:
**Describe the bug**

I typed "cmake .." as what is instructed by the INSTALL.md file to compile，then err occured: "Could NOT find ODE" , the platform is Ubuntu 18.04 LTS in vmware workstation 15 pro

**To Reproduce**
Steps to reproduce the behavior:
$ cd /path/to/grsim_ws
$ git clone https://github.com/RoboCup-SSL/grSim.git
$ cd grSim
$ mkdir build
$ cd build
$ cmake ..

**Expected behavior**
compile successfully

**Screenshots**
Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find ODE (missing: ODE_LIBRARIES ODE_PRECISION) (found version
  "")
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindODE.cmake:32 (find_package_handle_standard_args)
  CMakeLists.txt:72 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/drone/Desktop/grSim/build/CMakeFiles/CMakeOutput.log".

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**

kkimurak, 2020-03-26:
It would be a problem that missing library "libode-dev", as error message tells.  
In short, please follow the step :

1. Install `ode` with:
```
$ sudo apt install libode-dev
```

2. Remove grSim/build/CMakeCache.txt (or simply delete build directory)
```sh
$ cd /path/to/grsim/you/cloned
$ rm build/CMakeCache.txt
# or simply remove build directory with: rm -r build
```

3. Compile again
```
# create build directory if you deleted build directory in step 2
$ cd build
$ cmake ..
```

If you face similar situation, please check "Dependencies" section in `INSTALL.md`.

SaltedfishLZX, 2020-04-10:
> It would be a problem that missing library "libode-dev", as error message tells.
> In short, please follow the step :
> 
>     1. Install `ode` with:
> 
> 
> ```
> $ sudo apt install libode-dev
> ```
> 
>     1. Remove grSim/build/CMakeCache.txt (or simply delete build directory)
> 
> 
> ```shell
> $ cd /path/to/grsim/you/cloned
> $ rm build/CMakeCache.txt
> # or simply remove build directory with: rm -r build
> ```
> 
>     1. Compile again
> 
> 
> ```
> # create build directory if you deleted build directory in step 2
> $ cd build
> $ cmake ..
> ```
> 
> If you face similar situation, please check "Dependencies" section in `INSTALL.md`.
I met the same problems(on 18.04) even though I have install libode(use "apt-get" and compile ODE from source,all failed). Since I had install grSim successfully months ago,I search the commit log and revert to the version before "Improve FindODE.cmake"(it's where the error occurs).And it works.
Maybe this improvement makes some new bugs,I guess.

SaltedfishLZX, 2020-04-10:
> It would be a problem that missing library "libode-dev", as error message tells.
> In short, please follow the step :
> 
> Install ode with:
> 
> $ sudo apt install libode-dev
> 
> 
> Remove grSim/build/CMakeCache.txt (or simply delete build directory)
> 
> $ cd /path/to/grsim/you/cloned
> $ rm build/CMakeCache.txt
> # or simply remove build directory with: rm -r build
> 
> Compile again
> 
> # create build directory if you deleted build directory in step 2
> $ cd build
> $ cmake ..
> 
> If you face similar situation, please check "Dependencies" section in INSTALL.md.

I met the same problems(on 18.04) even though I have installed libode(use "apt-get" and compile ODE from source,both failed). Since I had install grSim successfully months ago,I search the commit log and revert to the version before "Improve FindODE.cmake"(it's where the error occurs).And it works.
Maybe this improvement makes some new bugs,I guess.

SaltedfishLZX, 2020-04-10:
> It would be a problem that missing library "libode-dev", as error message tells.
> In short, please follow the step :
>
> Install ode with:
>
> $ sudo apt install libode-dev
>
>
> Remove [...]

I met the same problem even I have Install ode. I also try to compile it form source, and met the same problem. Finally I revert the code before the commit "Improve FindODE.cmake" ,and this time I install it successfully. Maybe some change in  FindODE.cmake makes this error(missing ODE_LIBRARIES ODE_PRECISION),not very sure.

lucasg1, 2020-04-10:
None of the solutions that I searched worked, so just use release version "v2.2": https://github.com/RoboCup-SSL/grSim/releases/tag/v2.2

kkimurak, 2020-04-11:
I've created new VM (Ubuntu 18.04 LTS) and tested just now, but compiling current master branch succeed without any issues.  

Ah, OK, not written in README.md and any other documents, but `pkg-config` is required to find out ODE. The commit you specified ([Improve FindODE.cmake](88ef540af0d849e859a545e98ea3252afac31a6d) made this change.  

Could you try install `pkg-config` and build again?
It it fixes the problem, We need to update documents.

SaltedfishLZX, 2020-04-11:
> I've created new VM (Ubuntu 18.04 LTS) and tested just now, but compiling current master branch succeed without any issues.
> Ah, OK, not written in README.md and any other documents, but pkg-config is required to find out ODE. The commit you specified (Improve FindODE.cmake) made this change.
> Could you try install pkg-config and build again?
> It it fixes the problem, We need to update documents.

finally sovle this error after installing pkg-config~ thanks!

closed 2020-05-01 by mahi97