RoboCup-SSL/grSim #110: Whether the simulator can obtain images with distortion obtained by two global cameras

Droliven, 2020-03-25:
Does this simulator directly simulatethe 3D model ?
How could I get the raw imgs with  distortion  from two global cameras?
On this basis, I want to study distortion correction and camera calibration.

mahi97, 2020-05-01:
If I understand your question correctly, I should say Grsim does not simulate a camera and distortion of images.
It uses the ODE physics engine on the backend, and output comes directly from ode, and with some tricks, it looks like multiple cameras with distortion and errors.
The gui is just and close representation of the 3D world, which you can explore.

mahi97, 2020-05-01:
If you want to capture the OpenGL video of the simulator I have a note from @mani-monaj that may help you.
```
We are using the information and tools from glc-project

https://github.com/nullkey/glc

*** INSTALLATION (https://github.com/nullkey/glc/wiki/Install)

1. Install dependencies:

sudo apt-get install build-essential cmake libx11-dev libxxf86vm-dev libgl1-mesa-dev libasound2-dev libpng12-dev mplayer mencoder lame

2. Grab install script

http://github.com/nullkey/glc/blob/master/scripts/glc-build.sh

3. chmod a+x glc-build.sh 
4. run glc-build.sh

When asked about using git -> yes

*** USAGE (https://github.com/nullkey/glc/wiki/Capture)

1. goto Simulators obj dir
2. glc-capture --fps=30 --disable-audio ./Simulator 
3. Press shift-f8 to start/stop capturing

The resulting file will be Simulator-XXXX-Y.glc [VERY BIG]

*** PLAY (https://github.com/nullkey/glc/wiki/Playback)

glc-play Simulator-XXXX-Y.glc -y 1 -o - | mplayer -demuxer y4m -

*** ENCODE to mp4

glc-play ./Simulator-XXXX-Y.glc -o - -y 1 | mencoder -demuxer y4m - -nosound -ovc x264 -x264encopts qp=18:pass=1 -of avi -o video.avi

Have fun :)

-- Mani

```