RoboCup-SSL/grSim #113: Speedup Simulation time

goncamateus, 2020-04-30:
Hi guys! I'm trying to use grSim to train some Reinforcement Learning algorithms.
The whole environment is already set but I'm stucked with the simulation time problem.
grSim runs in real time and I want it to be something like 30+ times faster. 
Is there a way to do this? Have someone already tried?

Thank you in advance!

mahi97, 2020-05-01:
Hi
There are some configs under `Physics/World` that you can find in the configuration in gui.

`Desired FPS` shows how many loop you want the simulator takes in one second.
`ODE time step` shows how many seconds the simulator goes forward.
so for realtime simulation, the product of these numbers should be 1.

you can tune these numbers to get the product to 30, but there are drawbacks.
if you increase the `Desired FPS` it consumes more resources and you may not have enough.
if you increase the `ODE time step` it reduces the accuracy of the simulator.

I suggest to also use the `headless` mode by running grsim with additional `--headless` or `-h` argument to help the speed.

mahi97, 2020-05-01:
Also if you need more speed and don't care about the noise you can replace this line:
https://github.com/RoboCup-SSL/grSim/blob/a2a407d9785106d27876304099556ab862234d01/src/physics/pworld.cpp#L171
With `dWorldQuickStep` function.

I guarantee that this is not working well, but it runs way too faster. 
Improve performance by 50%.
Usually, `headless` mode improves performance by 15%.
So I suggest to combine them if you need more speed.

Another point that I missed, If you increase the `ODE time step` too much, your robots can pass through walls or other robots at high speeds. 
so be careful with it :wink:

goncamateus, 2020-05-01:
Nice! Thank you very much! I'm gonna try that!

closed 2020-05-01 by goncamateus