RoboCup-SSL/grSim #125: CI and CodeFactor

kkimurak, 2020-09-25:
First, As you may already know, Travis-CI has announced that they will close travis-ci.org and integrate it into travis-ci.com (the deadline is December 31, 2020).
https://mailchi.mp/3d439eeb1098/travis-ciorg-is-moving-to-travis-cicom
I think the project owner or maintainer need to do some migration operation.

Second, the CodeFactor in the README.md is linked to the ParsianRoboticLab/grSim, so it hasn't been updated for at least a year.
This will also require operation by the owner or maintainer.

mahi97, 2020-09-30:
I don't have enough permission to fix Travis and Codefactor since the repository is in the RoboCup-SSL organization.
I send a request to get the permission for Travis and Codefactor.

I think@g3force  can grant this permission to me to fix and keep maintaining this project.

g3force, 2020-09-30:
Hi @mahi97, I've accepted the two requests.
You already have the admin role for the grSim project, but new third party service integrations must first be approved for the organization, before they can be used in a project, that's why you could not approve it yourself.

kkimurak, 2020-09-30:
Thank you for your prompt response. Please close this issue once the settings have been corrected.

g3force, 2021-01-17:
@mahi97 what needs to be done to fix this issue?

mahi97, 2021-01-18:
I still can not add Robocup-ssl/grsim in the codefactor. The root cause is either my role in the organization or the codefactor integration with robocup-ssl. 

Can you try to see if the project is available in your codefactor account?

mahi97, 2021-01-18:
For the CI part, the migration is a little time consuming and unfortunately, I couldn't make time for reading the documents and start the migration.

I hope we can rely on the SSL community for this change too, in the end, I will check to see whether the CI is consistent with installation scripts and other platforms.

g3force, 2021-01-18:
Now it seems to work (after I have logged in to code factor and grant permission again): https://www.codefactor.io/repository/github/robocup-ssl/grsim/overview/master

mahi97, 2021-01-19:
Yeah it's working, I fix the configuration and update the badge. 👍

kkimurak, 2021-02-01:
close thanks to #141 . I'll update title to clarify that this issue was about badge url and migration.

closed 2021-02-01 by kkimurak