RoboCup-SSL/grSim #126: GrSim on Arch User Linux Repositories

Bollos00, 2020-09-27:
Good night,

I have created a [new package of grSim on the AUR](https://aur.archlinux.org/packages/grsim-git/). You can add on the [install instructions file](https://github.com/RoboCup-SSL/grSim/blob/master/INSTALL.md) if you want to. I think it is a good thing since there is not an instruction on how to install grSim on Arch Linux based distributions. It would be something like this:

-----------------------
## Arch Linux or derivative
[grsim-git](https://aur.archlinux.org/packages/grsim-git/) is available on the AUR, you can install it with
```bash
yay -S grsim-git
```
-----------

Thanks in advance.

mahi97, 2020-09-30:
Well Done,
It's nice to have this easy installation.
It would be great if we can have this kind of package for the Debian base too.

Please feel free to send a PR and add it to INSTALL.md

kkimurak, 2020-09-30:
Great work!

BTW, grSim requires `double-precision` enabled 'ode' (at least README.md and INSTALL.md say so). This package seems to be installing ode without `double-precision`.
Does this package works as expected (I mean, works without segmentation-fault and so on)?

If this works, non-Debian users also able to use grSim without building ode from source.

I'm setting up Arch Linux but it takes a time. I'm glad if you have answer.

Bollos00, 2020-09-30:
@mahi97 
Yes, I think that doing a release of a .deb package would be helpful for most users since the majority uses Debian or Ubuntu. I will do a pull request as soon as possible.

Bollos00, 2020-09-30:
@kkimurak 
I have been using grSim for two or three months and I have never installed `ode` from the source code, I have just installed [ode](https://www.archlinux.org/packages/community/x86_64/ode/) from the official repositories and it worked fine, I have never come across any issue. When I used ubuntu I also never needed to install it from the source code.

The note on the INSTALL.md file just says it is necessary to configure with the option `--enable-double-precision` **if you are installing it from the source code**.

kkimurak, 2020-09-30:
> The note on the INSTALL.md file just says it is necessary to configure with the option --enable-double-precision if you are installing it from the source code.

Oops, You're right, I was totally confused.
What I was looking at was [build configuration for package `ode-011.git`](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ode-011)  (listed in [`grsim-git`](https://aur.archlinux.org/packages/grsim-git/) dependencies), it is built without `double-precision`.
Package `ode` (in community repo, not `ode-011.git` in AUR), [is built `double-precision` enabled by default](https://github.com/archlinux/svntogit-community/blob/cecf0c157f39e6a830bab052db2565240937e031/ode/trunk/PKGBUILD#L29).

Finally, `grsim-git` worked perfectly in my Arch Linux VM. Thank you again for your great work (and sorry for misunderstood question).

Bollos00, 2020-09-30:
@kkimurak that is nothing and thanks for testing, I was kinda afraid the package would only work on my PC, especially because I am using Manjaro. Since you installed it from a fresh Arch Linux machine, all the dependencies are probably corrected.

PR #127 (2020-10-14): Install instructions for arch linux and desktop shortcup updated
Fixes #126 
### Description of the Change

I have created a new package of grSim on the AUR, as quoted [here](https://github.com/RoboCup-SSL/grSim/issues/126). It seems to be working fine, so I have updated the INSTALL instructions file. I have changed a little the structure of  the part 'Linux/Unix installation' and have also added the install instruction on the end, please see if you agree with the changes that were made.

I have also updated the Desktop entry file.

### Release Notes

Install instructions for Arch Linux system [... cut]
closed 2020-10-14 by mahi97