RoboCup-SSL/grSim #140: Windows support and installation instructions

bucefalog, 2021-01-30:
**Is your feature request related to a problem? Please describe.**
Many (most) of our team members use windows as their main OS. 
It would be great if grSim could compile in windows out of the box as well. 

**Describe the solution you'd like**
If this is a feature that could be added to the main repo, we could provide a fork (https://github.com/bucefalog/grSim/tree/windows) with updated install instructions, CMakeLists and necessary updates to the code (4 updated lines in total).

Either way, thanks in advance!

g3force, 2021-01-30:
Sounds helpful. Can you create a PR for this?

I also wonder if we could have a travis-ci build for Windows. Looks like there is some early stage support: https://docs.travis-ci.com/user/reference/windows/
Can you try this out as well?

bucefalog, 2021-02-01:
I created the pull request. I have not worked with travis-ci before, but I had a look and it does seem possible. I will  try it out and report results as soon as I have some extra time (possibly this weekend).

g3force, 2021-02-01:
Might got obsolete, see #145

bucefalog, 2021-02-04:
I see it seems like you'll get credits from travis-ci, I'll wait until issue 145 is closed before attempting a build with travis-ci.

bucefalog, 2021-02-22:
OK, I see issue 145 was closed. 
I went ahead and modified the build github action file so that it now supports windows x64. The first build will take about 4hs since it will build the dependencies and store them in a cache. Following builds will skip this step and instead restore them from the cache, running in about 10-15mins.
I already created a pull request with the changes (issue 151). I had to include some extra modifications so that it would compile in windows once more after recent changes to the code.

mahi97, 2021-02-22:
Thanks, Merged!

closed 2021-02-22 by mahi97