RoboCup-SSL/grSim #145: Travis has run out of credits

g3force, 2021-02-01:
I just migrated the repository from travis-ci.org to travis-ci.com, as resources on travis-ci.org got really sparse and jobs took for ever. Just to notice now, that travis-ci.com offers a one-time credit balance of 1000 minutes that were quickly used up -.-

Now, we could either try to get some OSS credits or we need to migrate to another service, like circle-ci, which we use for others repositories in the RoboCup-SSL organization. GitHub actions would also be an alternative.

Any opinions?

mahi97, 2021-02-02:
Yeah, I just saw the new plan for Travis CI. It seems they don't provide unlimited free service for open-source anymore.
I believe both Circle-CI and Github Action provide support for Linux, Windows, osx. So either way, it's better to switch from Travis.

 Also Related to #140.

mahi97, 2021-02-02:
Personally, I would prefer circle ci; it also keeps integrity between SSL projects. But I'm afraid we might need to switch to another. Platform as if they are more likely to remove free plans too in the future.

g3force, 2021-02-02:
With CircleCI, free plan gives us 2500 credits per week, so 10.000 per month.
The smallest docker server consumes 5 credits per minute, smallest VM 10 credits.
So we get about 1000 build minutes a month.
https://circleci.com/pricing/

GitHub Actions gives us 2000 minutes per month. Windows and macOS builds have a multiplier of 2 and 10.
https://docs.github.com/en/github/setting-up-and-managing-billing-and-payments-on-github/about-billing-for-github-actions

As grSim itself is consuming quite a lot of build time, I'd suggest we'll try GitHub actions and save us the build minutes on circleci for the other projects.
We should also consider if we really need to have 6 different builds. Do we really need two different compilers? Building on different OS is really helpful. We can think about only supporting Ubuntu 18.04 or only 20.04, if we need to save more build time.

g3force, 2021-02-02:
We should also have a look at drone.io
They write that they are 100% free for open source. I can not find any details currently, though.

EtienneSchmitz, 2021-02-02:
Circle.ci has open source plans (https://circleci.com/open-source/).
In this page, they write : 
> Organizations on our free plan get 400,000 credits per month for open source builds.

It seems the same for Travis : https://www.travis-ci.com/plans
> Free for open source. We love the Open Source Community, and to show how much we love it, upon validated request placed with our Support Team you may receive free OSS credits for your public builds.

g3force, 2021-02-02:
Yeah, both need to be requested, though. We can give it a try.

g3force, 2021-02-04:
FYI: I've contacted the Travis-ci Support and it looks like they will sponsor us credits for free.

mahi97, 2021-02-05:
Yay, That's great.

g3force, 2021-02-13:
I have not heard back from the Travis-Support yet...

I have setup GitHub Actions now instead. Interestingly, for public repositories, GitHub Actions are unlimited :) And they support MacOS and Windows as well.

PR #149 (2021-02-13): Add github actions workflow
closes #145
closed 2021-02-13 by mahi97