RoboCup-SSL/grSim #153: `grSim_RobotReplacement::yellowteam` is not used to determine the team for RobotReplacement

YuseiNaito, 2021-02-25:
**Describe the bug**
When we try to reposition the **yellow** robot according to `grSim_Replacement.proto`, the **blue** robot will be repositioned.

**To Reproduce**
Steps to reproduce the behavior:
1. In our case of C++
    ```c++
    ssl_protos::grsim::Packet packet{}
    auto r = packet.mutable_replacement()->add_robots();
    r->set_id(id);
    r->set_x(x);
    r->set_y(y);
    r->set_dir(dir);
    r->set_yellowteam(true);

    connection_->send(packet.SerializeAsString());
    ```
2. The **blue** `id` robot will be repositioned .

**Expected behavior**
Robots should be repositioned according to `grSim_RobotReplacement::yellowteam`

**Desktop (please complete the following information):**
 - OS: Ubuntu 
 - Version 20.04.2 LTS

**Additional context**
With [this change](https://github.com/RoboCup-SSL/grSim/commit/1c97f6b1e1fff1c7c5eed4aac7b8a78d3fee1510#diff-8f241c263325a3bb6f353258373a7a7035f03c0f1118bc4e7e4138f4fd2029efL607-R584), teams are now judged by `grSim_commannds::isteamyellow` instead of `grSim_RobotReplacement::yellowteam` when RobotReplacement.
If we set `grSim_commannds::isteamyellow`, we can do RobotReplacement correctly.

EtienneSchmitz, 2021-03-08:
Same problem is our team, do you want to do a pull request ?  @YuseiNaito

g3force, 2021-03-09:
I wonder why nobody has noticed this before... =D

A PR would be nice. This should be fixed, although there is the new standard protocol now also, so you could just migrate to this one.

YuseiNaito, 2021-03-10:
Thanks. I created PR to fix this.

mahi97, 2021-03-10:
Thanks, merged.

closed 2021-03-10 by mahi97