RoboCup-SSL/grSim #160: Possible division by zero on robot speed handling causing crashes

FelipeMartins96, 2021-04-20:
https://github.com/RoboCup-SSL/grSim/blob/206c8dcb0b354325c1e0542635f3ae1c71c0c669/src/robot.cpp#L472-L473

How to reproduce:
Send v_x and v_y zero to a robot with current speed > AccBrakeAbsoluteMax * delta_t

g3force, 2021-04-20:
oh, sry... I've missed that. :facepalm: 

I can prepare a patch in the next days or if you like, you can also provide a patch yourself via Pull Request.

FelipeMartins96, 2021-04-22:
I`ve tried some solutions, but they haven`t worked so far, I'll try testing other ideas later

g3force, 2021-04-23:
Could you test the following fix?
```
        if (v > 0) {
            vx *= new_v / v;
            vy *= new_v / v;
        } else {
            vx = cvv[0] * (new_v / cv);
            vy = cvv[1] * (new_v / cv);
        }
```

FelipeMartins96, 2021-04-23:
it didn't work properly, I think it is because cvv is in global coordinates while vx and vy are local, I'll try converting it

FelipeMartins96, 2021-04-23:
```
        if (v > 0) {
            vx *= new_v / v;
            vy *= new_v / v;
        } else {
            // convert global to local
            dReal k, angle;
            angle = getDir(k);
            angle *= _DEG2RAD;
            dReal cvx = cvv[0]*cos(angle) + cvv[1]*sin(angle);
            dReal cvy = -cvv[0]*sin(angle) + cvv[1]*cos(angle);

            vx = cvx * (new_v / cv);
            vy = cvy * (new_v / cv);
        }
```
I think this fix works