RoboCup-SSL/grSim #167: Software for Manjaro Linux

Jyo561, 2021-09-30:
Please provide a documentation on how to install and run it in Manjaro Linux

g3force, 2021-10-01:
Have you tried the AUR package from @Bollos00 ?
https://aur.archlinux.org/packages/grsim-git/

If you do not want to install it directly, you can consult the PKGBUILD file for the required dependencies.

Bollos00, 2021-10-01:
Yes. Also, the install instructions to install it on Arch Linux is exactly the same to install it on manjaro. https://github.com/RoboCup-SSL/grSim/blob/master/INSTALL.md#arch-linux

Jyo561, 2021-10-19:
> Yes. Also, the install instructions to install it on Arch Linux is exactly the same to install it on manjaro. https://github.com/RoboCup-SSL/grSim/blob/master/INSTALL.md#arch-linux

I tried it but it says to use another tool as cmake is unable to find the build tool

Jyo561, 2021-10-19:
i installed it again and ran cmake command given in the installation guide but still it shows this error:-

CMake Error at /var/lib/snapd/snap/cmake/936/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find ODE (missing: ODE_LIBRARIES ODE_INCLUDE_DIR ODE_PRECISION)
  (found version "")
Call Stack (most recent call first):
  /var/lib/snapd/snap/cmake/936/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindODE.cmake:32 (find_package_handle_standard_args)
  CMakeLists.txt:76 (find_package)

Bollos00, 2021-10-19:
The error message says you do not have ODE installed, so probably you did not installed the packages that are on the second line of the command...

Try this:
```
sudo pacman -S base-devel boost hicolor-icon-theme mesa ode protobuf qt5-base cmake git
```

closed 2021-10-19 by Jyo561