RoboCup-SSL/grSim #173: cmake in Mac OS

mohammadhosseinz, 2022-05-25:
maxOS Monterey 12.2.1
cmake 2.23.1

**cant cmake **
CMake Error at CMakeLists.txt:68 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

kkimurak, 2022-05-30:
Could you try `brew install qt@5` to install and re-run `cmake` ?

We need `qt` 5.x but the formula `qt` is now 6.x (the upgrade was introduced on 3 March 2021, by commit https://github.com/Homebrew/homebrew-core/commit/ab9d906d072fb2c5ba6ca30b23cb3136e42de6c1 . Current version is [6.3.0](https://formulae.brew.sh/formula/qt) when I checked.). So we should use `qt@5` alias to install qt version 5.x.

mohammadhosseinz, 2022-05-30:
@kkimurak 

> Could you try `brew install qt@5` to install and re-run `cmake` ?
> 
> We need `qt` 5.x but the formula `qt` is now 6.x (the upgrade was introduced on 3 March 2021, by commit [Homebrew/homebrew-core@ab9d906](https://github.com/Homebrew/homebrew-core/commit/ab9d906d072fb2c5ba6ca30b23cb3136e42de6c1) . Current version is [6.3.0](https://formulae.brew.sh/formula/qt) when I checked.). So we should use `qt@5` alias to install qt version 5.x.

mohammadhosseinz, 2022-05-30:
tank for your attention
I use this command:
```
brew install qt@5
brew uninstall qt
brew uninstall cmake
brew install cmake
```
but is the same error.
@kkimurak

mohammadhosseinz, 2022-06-21:
Thank you it worked.

closed 2022-06-21 by mohammadhosseinz

AlexM-dev, 2023-08-16:
> OK, I have noticed that we need some more trick. Run
> 
> ```shell
> $ export Qt5_DIR=$(brew --prefix qt@5)/lib/cmake/Qt5
> ```
> 
> then build as usual. I have confirmed it works on Mojave but I believe it works on Monterey as well.
> 
> Additional note: I have searched more and found that brew-installed qt is not linked by default. So We set `CMAKE_PREFIX_PATH` [in `CMakeLists.txt`](https://github.com/RoboCup-SSL/grSim/blob/4cf460dd342a94187aeb0b92d8f27d8023f784e1/CMakeLists.txt#L61-L67) if platform is MacOS and `/usr/local/opt/qt` is available. That was a workaround for old Qt5 installation. With the change of qt version to 6.x, this installation directory has changed as well. That's why the `cmake` fail.
> 
> Even we change the line `list(APPEND CMAKE_PREFIX_PATH "/usr/local/opt/qt")` to `list(APPEND CMAKE_PREFIX_PATH "/usr/local/opt/qt5")`, the build may fail. vartypes (dependency of grSim) is also depends on qt5 and it will be built separately to grSim. Passing `CMAKE_PREFIX_PATH` to [`VARTYPES_CMAKE_ARG`](https://github.com/RoboCup-SSL/grSim/blob/4cf460dd342a94187aeb0b92d8f27d8023f784e1/CMakeLists.txt#L96) solves the problem.
> 
> In CI, those problems are avoided by [setting environment variable `Qt5_DIR` directly](https://github.com/RoboCup-SSL/grSim/blob/4cf460dd342a94187aeb0b92d8f27d8023f784e1/.github/workflows/build.yaml#L34-L35). This is another approach (but requires additional action by user). I'm not sure which is better.

I tried to do this, but get errors like 
`error: static_assert failed due to requirement '201103L >= 201402L' "Protobuf only supports C++14 and newer."
static_assert(PROTOBUF_CPLUSPLUS_MIN(201402L), "Protobuf only supports C++14 and newer.");
^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`
and 
`error: no type named 'in_place_t' in namespace 'absl'
  explicit constexpr Storage(absl::in_place_t, V&& v)
`

As I know I should use c++17 to fix this, I make changes in CMakeLists.txt, but this dont help