RoboCup-SSL/grSim #176: Cant build on Mac

AlexM-dev, 2023-08-15:
**Describe the bug**
I try to build grSim, but get errors like:
"In file included from /opt/homebrew/include/absl/types/span.h:69:
/opt/homebrew/include/absl/types/internal/span.h:130:24: error: no template named 'enable_if_t' in namespace 'std'; did you mean simply 'enable_if_t'?
using EnableIfIsView = std::enable_if_t<IsView<T>::value, int>;"
"/opt/homebrew/include/absl/container/internal/compressed_tuple.h:155:48: error: no type named 'in_place_t' in namespace 'absl'
  explicit constexpr CompressedTupleImpl(absl::in_place_t, Vs&&... args)"
etc.

**Expected behavior**
I think that to solve the problem I need to change the c++ version when building, but I don't understand how to do it

**Desktop**
 - OS: Mac OS

AlexM-dev, 2023-08-16:
I tried to add
`# set explicitly the c++ standard to use
set(CMAKE_CXX_STANDARD 17)`
in CMakeLists.txt but it dont help

Dont understand why

AlexM-dev, 2023-08-18:
I fix this. Now I'm stuck on an error when "make": "Undefined symbols for architecture arm64".

Leikoe, 2023-12-25:
my build fails at ld, Undefined symbols
```void absl::lts_20230802::log_internal::LogMessage::CopyToEncodedBuffer<(absl::lts_20230802::log_internal::LogMessage::StringType)0>(std::__1::basic_string_view<char, std::__1::char_traits<char>>),```

closed 2024-02-24 by g3force